Snapmaker/Luban #1770: Bug: in Github usage. SET ISSUES TO CLOSED or TAG or ADD COMMENT

HorstBaerbel, 2022-11-30:
Me, and probably other people here, are willing to test if bugs are actually solved, but we need to know they are. The changelog in new versions is NOT adequate for that and the GitHub issue list is overflowing. You seem to have an internal bug tracking / ticketing system at Snapmaker. My suggestion is to copy the link of the corresponding GitHub issue to that ticket system and either:

- Set the GitHub issue to closed (with a comment stating the version number it is solved in)
- Add a tag (Solved in version X.Y.Z)
- At the very least comment "Solved in version X.Y.Z, needs retesting" so people know.

zeroClearAmerican, 2023-01-01:
I second this. Willing to test and help out as much as I can but without sufficient communication/organization from the Snapmaker side translating into this repo's backlog of issues, we might as well be yelling at a brick wall.

zeroClearAmerican, 2023-03-25:
lol

HorstBaerbel, 2023-04-15:
Thanks for doing this. Much appreciated!