Snapmaker/Luban #1771: Bug: Stuck infinitely after homing in 3d printing

HorstBaerbel, 2022-11-30:
# 🐞 bug report

### Affected Version(s)
4.4.0

### Is this a regression? (optional)
Yes, regression. No problem in 4.3.2

### To Reproduce
Steps to reproduce the behavior:
1. Go to workspace
2. Luban wants to home printer, click ok
3. Printer homes itself correctly
-> Homing dialog stuck infinitely. Can't press Ok.

[Video](https://user-images.githubusercontent.com/1310747/204757682-83c53e91-e947-48e4-b28c-55568504df78.webm)

Expected behavior
Homing ends in finite time

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: A350T

andrewpierson, 2022-12-17:
Confirmed on Windows 11, A350.  Worked for me previously, as well on 4.3.2.  Closing the app and restarting it is an annoying workaround.

tensile23, 2023-04-04:
Should have been fixed in 4.7.2.

closed 2023-04-04 by tensile23

HorstBaerbel, 2023-04-15:
Fixed in 4.7.2