Snapmaker/Luban #1782: Bug: A150 strange behaviour in 3d printing

Airmonkdev, 2022-12-05:
# 🐞 bug report

### Affected Version(s)
4.4.0

### Is this a regression? (optional)

Yes, no problem on v3.15.2

### To Reproduce
Steps to reproduce the behavior:
Generate gcode (fast print, 100% infill) from the given stl file and print.

Expected behavior
Print without interruptions.

## 🌍  Your Environment

- Operating System: macOS Ventura (Apple M1)
- Printer: Snapmaker A150

</code></pre>

https://user-images.githubusercontent.com/74402751/205582716-71e8af22-8766-45c4-9b9c-59203936464f.mp4

[stl and gcode.zip](https://github.com/Snapmaker/Luban/files/10151849/stl.and.gcode.zip)

Andygol, 2022-12-05:
This is not a bug, but a feature. See for details here

https://github.com/wen-qing-255/ProfileDocs/blob/a59adebc9601cf6375320ec772666a47e3d827ae/EN/heating_and_cooling/cool_min_layer_time.md?plain=1#L1-L13

Andygol, 2022-12-11:
@Airmonkdev I assume that this setting might help in your case

[image]

Airmonkdev, 2022-12-12:
Thank you so much for the clarification.
My only concern is: in my case this feature doesn't work well because when the head is waiting, the filament is running out (don't know why!) so when it resume printing it cause a jam on the object.
Is it my fault? Am I wrong on some settings?

Thanks!