Snapmaker/Luban #1794: Bug: Quit and the entire Luban menu in Workspace not available

jenkins007, 2022-12-07:
# 🐞 bug report

### Affected Version(s)

 I am using 4.4.0

### Is this a regression? (optional)

 I don't think this has ever worked 

### To Reproduce
Steps to reproduce the behavior:
1. Go to Workspace
2. Click on Snapmaker Luban on the top menu bar
3. Note the entire drop down menu is grayed out and not available
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior
<!-– The menu should always be available and functional like all other apps. To access this drop down menu where it is functional, go back to Home and it is functional there-->

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [macOS Ventura with latest update (13.0.1)]
- Printer: [Snapmaker T350]

</code></pre>
[image]

parachvte, 2022-12-19:
Menu is disabled in workspace and other popup windows.

This is a temporary solution to prevent unwanted behavior and operations in pages other than the scene.
We will fix this soon.

parachvte, 2023-01-11:
#1863 Re-opens the menu.

closed 2023-01-11 by parachvte