Snapmaker/Luban #1815: Bug: xxxx in 3d printing

andrewpierson, 2022-12-17:
# 🐞 bug report

### Affected Version(s)
Only seen in 4.4.0

### Is this a regression? (optional)
Have not seen this in previous versions.

### To Reproduce
1. Connect Laser Module
2. Power on the machine
3. Connect with Luban
4. After homing. restart Luban due to failure to finish homing bug.
5. Load GCode from Lightburn.

Exception or Error (optional)
Inconsistent job type warning

Expected behavior
GCode runs as expected as I've seen previously.

## 🌍  Your Environment

**Platform:**
<pre><code>
- Operating System: Windows 11 x64
- Printer: A350
</code></pre>
[image]

sumibin, 2023-03-14:
If laser processing is required, rename the resulting Gcode to a .nc suffix.

closed 2023-03-14 by sumibin

andrewpierson, 2023-03-14:
While it is your prerogative to close this, it would be a better customer experience if the file types that are supported by the software are not rejected incorrectly.  I think it makes more sense if you fixed your product rather than forcing all of us to work around your bugs.