Snapmaker/Luban #1819: Bug: xxxx in 3d printing

Galindorf, 2022-12-19:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.4.0

### Is this a regression? (optional)

 unknown/can't remember

### To Reproduce
Steps to reproduce the behavior:
Load Laser project to workspace
choose "start on Luban, input material thickness
Job is sent to laser
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

My project was having an issue and there was no way to stop the laser except to open the door. There is no pause or stop anywhere in Luban, and no way to continue a paused job. After opening the door, I affixed my laser material properly (because it always cuts the outline first, very frustrating) and the job would not resume after I closed the door. The connected laser shows a 'paused' icon with no way to resume job except to start it over. 

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10
- Printer: Snapmaker A350 10w laserhead

</code></pre>

[image]

trgall, 2022-12-23:
Click on the small up arrow next to "Job Status" in the upper right hand corner of the screen.  This will pull down the Job Status that includes a Pause and Stop button.

I don't know why this is collapsed as a default.  The default should be to always show this, including when there is no job with "No Job" listed and with the buttons grey.

Galindorf, 2023-01-07:
Thank you for the advice!
There is another issue where the "job status" panel does not immediately appear when a job has begun. It really should be available at all times in my opinion~

closed 2023-01-07 by Galindorf