Snapmaker/Luban #1827: Bug: Slicer do not work in 3d printing for 4.5.0-beta.8

Andygol, 2022-12-25:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.5.0-beta.8 with Snapmaker 2.0 A150

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 4.5.0-beta.6 or 4.5.0-beta.5

### To Reproduce
Steps to reproduce the behavior:
1. Clone the repo
2. nvm use 16
3. npm install && npm run dev
4. Put your *.stl on the table
5.  Click "Generate G-code… and nothing is going on, no any logs in Terminal.
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

Slicer is working

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: macOS Monterey 12.6.2 (M1)
- Printer: Snapmaker A150

</code></pre>

jburleson795, 2022-12-30:
I am having the same issue for the a350. the last version I checked that worked was 4.0.0 but stopped checking after I saw this one worked. 3d printing option is unusable for me currently.

parachvte, 2022-12-30:
Sorry for late response.

I think there are more than one issue here. 

@jburleson795 Please use latest version release (v4.4.0 instead of beta versions), I think they should work for you.

@Andygol Sorry for the inconvenience. We only changed the build platform from macOS 12 to macOS 11, with older frameworks (SDK) to rebuild the slicer. Which makes it runnable on macOS 11. So we will check if it's the binary not compatible with M1 chip (on next week).

By the meantime, could you help us to test the ARM64 build? Under https://github.com/parachvte/Luban/releases/tag/v4.5.0-beta.4, there is a test build name "[Snapmaker.Luban-4.5.0-beta.4-arm64.testing.dmg](https://github.com/parachvte/Luban/releases/download/v4.5.0-beta.4/Snapmaker.Luban-4.5.0-beta.4-arm64.testing.dmg)".

Andygol, 2022-12-31:
@parachvte I just tested beta-4 provided by you and tthe slicer works.

jburleson795, 2022-12-31:
Good morning,
In regards to the issue, I found that the problem wasn't the slicer. The
application was backlogging configurations and presets from every update
and reinstallation to a different location on my computer that was not
affected by uninstalling the app. Once I found and cleared this folder,
everything began working fine.
Thank you for your time
Jordan Burleson

parachvte, 2023-01-05:
@Andygol 

The slice issue should be fixed in [4.5.0-alpha.17](https://github.com/parachvte/Luban/releases/tag/v4.5.0-alpha.17)
Or `main` branch.

IgorC76, 2023-01-06:
Just downloaded 4.5.0-alpha.17 for use with Snapmaker J1. Running on Linux (6.0.15-300.fc37.x86_64). Slicer doesn't work. If i try to run LunarSlicer independently with the same parameters as luban executable, I get a bit more info:
....
Progress:inset+skin:55:100 	0.162825%
Progress:inset+skin:58:100 	0.170402%
Progress:inset+skin:64:100 	0.185555%
Progress:inset+skin:65:100 	0.188080%
Progress:inset+skin:75:100 	0.213336%
Progress:inset+skin:92:100 	0.256269%
Progress:inset+skin:2:2 	0.276473%
Layer count: 93
Progress: inset+skin accomplished in 0.125s
Starting support...
[ERROR] Trying to retrieve setting with no value given: 'fill_perimeter_gaps'

It seems that 'fill_perimeter_gaps' is not defined

Andygol, 2023-01-06:
> @Andygol
> 
> The slice issue should be fixed in [4.5.0-alpha.17](https://github.com/parachvte/Luban/releases/tag/v4.5.0-alpha.17) Or `main` branch.

There is no version for Apple Silicon 😢, but I downloaded https://github.com/parachvte/Luban/archive/refs/tags/v4.5.0-alpha.17.zip and tried to compile and run Luban by my own. Unfortunately it was unable even to open *.stl

[image]

parachvte, 2023-01-10:
@Andygol 

I added a test build for ARM64 here: [Snapmaker.Luban-4.5.0-beta.9-arm64.dmg](https://github.com/parachvte/Luban/releases/download/v4.5.0-m1.2/Snapmaker.Luban-4.5.0-beta.9-arm64.dmg)

Andygol, 2023-01-10:
> @Andygol 
> 
> I added a test build for ARM64 here: [Snapmaker.Luban-4.5.0-beta.9-arm64.dmg](https://github.com/parachvte/Luban/releases/download/v4.5.0-m1.2/Snapmaker.Luban-4.5.0-beta.9-arm64.dmg)
> 
> 
> 
> 

@parachvte thank you, the slicer works as expected. 

I was able to load *.stl and generate g-code and load it to Workspace.

parachvte, 2023-01-11:
> Just downloaded 4.5.0-alpha.17 for use with Snapmaker J1. Running on Linux (6.0.15-300.fc37.x86_64). Slicer doesn't work. If i try to run LunarSlicer independently with the same parameters as luban [...]

Sorry that linux version of slicer is not updated yet.

addiedavis, 2023-01-13:
I downloaded 4.5.0 release using Snapmaker 2.0 A350 and the 3D module. 
The gcode slices, but when the gcode is started either on Luban or on the Snapmacker the prints do not trigger anything on the snapmacker. It does not even start heating the nozzle or bed. 

Side note: older gcodes that were left in my snapmaker machine do print - so it was something in the Luban update yesterday (01/12/2023).

parachvte, 2023-01-13:
> The gcode slices, but when the gcode is started either on Luban or on the Snapmacker the prints do not trigger anything on the snapmacker. It does not even start heating the nozzle or bed.

Could you share your G-code, or a brief head & first few lines of G/M codes?

parachvte, 2023-01-16:
Close this issue.

For issue @addiedavis mentioned, use https://github.com/Snapmaker/Luban/issues/1878

closed 2023-01-16 by parachvte