Snapmaker/Luban #1830: Bug: The packaged executable file cannot be run

Timicasto, 2022-12-30:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.4.0 

### Is this a regression? (optional)

 no

### To Reproduce
Steps to reproduce the behavior:
1. clone the source code
2. nvm use 12
3. npm install -g npm@7.24.0
4. npm run build
5. npm run electron-builder
6. run the produced executable file 
7. see the error (in stdout)

this problem is caused by the sandbox system in electron, once I just disabled sandbox by `--no-sandbox` it just fixed. But I think this problem will affect more and more people who are not familiar to electron... so this should be disabled in the code (or disable it after platform judging, cuz I dunno whether this problem happens on older kernels) <br>

Exception or Error (optional)
<pre><code>
[2030262:1230/142028.653949:FATAL:gpu_data_manager_impl_private.cc(445)] GPU process isn't usable. Goodbye.
[1]    2030262 trace trap (core dumped)  /opt/Snapmaker\ Luban/snapmaker-luban
</code></pre>

Expected behavior
window is created and the program should startup normally

## 🌍  Your Environment

**Platform:**
<pre><code>

- OS:  Arch Linux x86_64
- Kernel: 6.1.1-native_intel-xanmod1-1 (Linux timicasto-main 6.1.1-native_intel-xanmod1-1 #1 SMP PREEMPT_DYNAMIC Fri, 23 Dec 2022 01:50:24 +0000 x86_64 GNU/Linux Linux version 6.1.1-native_intel-xanmod1-1 (makepkg@archlinux) (clang version 14.0.6, LLD 14.0.6) #1 SMP PREEMPT_DYNAMIC Fri, 23 Dec 2022 01:50:24 +0000)
- DE: Gnome 43.2
- WM: Mutter(X11)
- CPU: Intel Core i7-8086K (12) @ 5 GHz
- GPU: NVIDIA GeForce RTX 2070 Rev. a
- Memory: 31.29GB
- Printer: Snapmaker A350T (seems to be useless)

</code></pre>

Timicasto, 2022-12-30:
`# Maintainer: TurboPunk <turbopunk@telenet.be>
# Contributor: José San Juan <josesj@gmail.com>
pkgname=snapmaker-luban
pkgver=4.4.0
pkgrel=4
pkgdesc="Snapmaker Luban is an easy-to-use 3-in-1 software tailor-made for Snapmaker machines."
url="https://github.com/Snapmaker/Luban"
license=('AGPL v3')
arch=('x86_64')
makedepends=('npm' 'nvm' 'jq' 'moreutils')
options=('!strip')
source=("https://github.com/Snapmaker/Luban/archive/refs/tags/v${pkgver}.tar.gz")

sha256sums=('6db462e15fa55691226351a5a908b00b9b25b32aff1734dab2009091fe29a2ee')

_ensure_local_nvm() {
    # let's be sure we are starting clean
    which nvm >/dev/null 2>&1 && nvm deactivate && nvm unload
    export NVM_DIR="${srcdir}/.nvm"

    # The init script returns 3 if version specified
    # in ./.nvrc is not (yet) installed in $NVM_DIR
    # but nvm itself still gets loaded ok
    source /usr/share/nvm/init-nvm.sh || [[ $? != 1 ]]
}

prepare() {
  _ensure_local_nvm
  nvm install 12
  cd ${srcdir}/Luban-${pkgver}
  jq '.build.linux.target="pacman"' package.json | sponge package.json
}

build() {
  _ensure_local_nvm
  export PATH=$PATH:node_modules/.bin
  cd ${srcdir}/Luban-${pkgver}
  nvm use 12
  npm install --cache "${srcdir}/npm-cache" -g npm@7.24.0
  npm install --cache "${srcdir}/npm-cache"
  npm run build
  npm run electron-builder
}

package() {
  mv ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.pacman ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.tar.xz
  tar xf ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.tar.xz --exclude='.*' -C ${pkgdir}
  chown -R root:root "${pkgdir}"
}`

Timicasto, 2022-12-30:
```# Maintainer: TurboPunk <turbopunk@telenet.be>
# Contributor: José San Juan <josesj@gmail.com>
pkgname=snapmaker-luban
pkgver=4.4.0
pkgrel=4
pkgdesc="Snapmaker Luban is an easy-to-use 3-in-1 software tailor-made for Snapmaker machines."
url="https://github.com/Snapmaker/Luban"
license=('AGPL v3')
arch=('x86_64')
makedepends=('npm' 'nvm' 'jq' 'moreutils')
options=('!strip')
source=("https://github.com/Snapmaker/Luban/archive/refs/tags/v${pkgver}.tar.gz")

sha256sums=('6db462e15fa55691226351a5a908b00b9b25b32aff1734dab2009091fe29a2ee')

_ensure_local_nvm() {
    # let's be sure we are starting clean
    which nvm >/dev/null 2>&1 && nvm deactivate && nvm unload
    export NVM_DIR="${srcdir}/.nvm"

    # The init script returns 3 if version specified
    # in ./.nvrc is not (yet) installed in $NVM_DIR
    # but nvm itself still gets loaded ok
    source /usr/share/nvm/init-nvm.sh || [[ $? != 1 ]]
}

prepare() {
  _ensure_local_nvm
  nvm install 12
  cd ${srcdir}/Luban-${pkgver}
  jq '.build.linux.target="pacman"' package.json | sponge package.json
}

build() {
  _ensure_local_nvm
  export PATH=$PATH:node_modules/.bin
  cd ${srcdir}/Luban-${pkgver}
  nvm use 12
  npm install --cache "${srcdir}/npm-cache" -g npm@7.24.0
  npm install --cache "${srcdir}/npm-cache"
  npm run build
  npm run electron-builder
}

package() {
  mv ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.pacman ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.tar.xz
  tar xf ${srcdir}/Luban-${pkgver}/output/${pkgname}-${pkgver}.tar.xz --exclude='.*' -C ${pkgdir}
  chown -R root:root "${pkgdir}"
}```

parachvte, 2023-01-11:
We have a  pull request https://github.com/Snapmaker/Luban/pull/1868 to solve this.

@Timicasto Can you test if this branch works for you. Even better to check if only enable `--disable-setuid-sandbox` can solve this problem.

PR #1868 (2023-01-13): Fix: Fix linux executable by adding --no-sandbox option to electron
// Closes #1830 #1866
closed 2023-01-13 by parachvte