Snapmaker/Luban #1844: Bug: Can't continue to cut if enclosure door were open when starting in Laser

HorstBaerbel, 2023-01-05:
# 🐞 bug report

### Affected Version(s)
4.4.0

### Is this a regression? (optional)
Not sure

### To Reproduce
Steps to reproduce the behavior:
1. Have enclosure door detection enabled and open one door
2. Start laser cut in Luban
3. Head moves and when cutting should start Luban prompts to close door to continue
4. Close enclosure door -> Nothing happens, Popup message does not go away

[image]

Expected behavior
Luban continues cutting after closing doors

## 🌍  Your Environment

**Platform:**
- Operating System: Fedora 36
- Printer: Snapmaker A350T

Galindorf, 2023-01-07:
I think this is related to an issue I posted earlier:

https://github.com/Snapmaker/Luban/issues/1819

The problem I had is that I can only seem to pause a job on Luban BY opening the door... and then it won't start again. 
There is a setting that is basically HIDDEN on the the Workspace panel that appears after a job has started, called "Job Status" 
Here where I first began the job, the panel is missing. 
[image]
This panel did not appear until after the job was already begun and was almost 18% finished. If you experience a head crash or other emergency situation,  you pull open the door to stop it immediately.  Here you can see the "Job Status" panel has appeared, you have to open it to be able to pause/start the current job. 

[image]

I was able to pause and restart the laser both by pausing the job under Job Status, and by restarting it after opening the door. The trick is not obvious, you have to actually click the "X" on the warning panel to close it. On my version the warning panel does not go away after you close the door. 
[image]

I hope this is helpful!

HorstBaerbel, 2023-01-08:
Thanks. Then the warning panel seems to be missing a button. Either "Close" (close dialog, do nothing) or best "Continue" (greyed out if doors open, pushable if doors closed).