Snapmaker/Luban #1849: Bed Temperature from material is ignored in Luban 4.4.0

fhorlaville, 2023-01-07:
Hi,

I created a new material as I have a silk-metal PLA filament which needs a cold to 30º bed temperature.

I used quick create and set the bed temperature to 0-30. This saved properly and displays properly.

Unfortunately, the G-code ignores the setting and sets the bed to 70º to start with and then to 50º, completely ignoring the setting.

This is Luban 4.4.0 for Mac.

Thanks

**Platform:**
<pre><code>

- Operating System: macOS Ventura 13.1
- Printer: Snapmaker A250

</code></pre>

parachvte, 2023-01-18:
Please check if v4.5.0 fix the problem.

fhorlaville, 2023-01-19:
Fixed in 4.5.0 where all the settings are now available - well done and thank you!

closed 2023-01-19 by fhorlaville