Snapmaker/Luban #1875: Bug: The "reset" functionality disappeared from the 3d printing screen

leandersabel, 2023-01-12:
# 🐞 bug report
It is no longer possible to reset the parameters to their default values of the build in 3d printing profiles. 

### Affected Version(s)
This issue is caused by version 4.5.0

### Is this a regression? (optional)
Yes, the previous version in which this bug was not present was: 4.4.0. 
It also works again as expected after downgrading to 4.4.0

### To Reproduce
Steps to reproduce the behavior:
1. Go to 3d printing screen
2. Click on any profile, e.g. "fast print"
3. Change any parameter, e.g. "layer height" from 0.24 mm to 0.16 mm
4. Hover over "Fast Print" and click on the three dots.
5. Click "reset"

Exception or Error (optional)
In version 4.5.0, the "reset" functionality is no longer there. The only option available is copy

Expected behavior
After the click on "reset", all parameters should return to their defaults. 

## 🌍  Your Environment

**Platform:**
- macOS Ventura 13.0.1
- MacBook Pro 13-inc, 2019
- Printer: Snapmaker A350

parachvte, 2023-01-18:
The print settings dialog was refined in v4.5.0, but the "reset" functionality is not migrated to the new dialog for now.

We will add it back in next minor update.

sumibin, 2023-02-22:
Done in v4.6.2

closed 2023-02-22 by sumibin