Snapmaker/Luban #1877: Bug: Luban distorts SVG files when importing (Laser, CNC)

ProfCampbokk, 2023-01-12:
# 🐞 bug report

### Affected Version(s)
Visible in 4.4.0 and 4.5.0

### Is this a regression? (optional)

 I'm not aware of this working previously.

### To Reproduce
Steps to reproduce the behavior:
1. Go to 2D laser mode
2. Open my SVG file
3. Observe the distortion
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

N/A
</code></pre>

Expected behavior

The file should load identically to how it looks in other software, e.g. Lightburn or Inkscape.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [macOS 13.1 (22C65)]
- Printer: [Not relevant]

</code></pre>

[image]

https://user-images.githubusercontent.com/98917104/212184597-40b62e9a-26a2-4c7e-a9b3-b8f33d988a27.mov

ProfCampbokk, 2023-01-22:
Also mentioned in this issue: https://github.com/Snapmaker/Luban/issues/1884

parachvte, 2023-01-31:
This issue is different from #1884 

```
<tspan
           sodipodi:role="line"
           id="tspan1004"
           style="font-size:8.46667px;text-align:center;text-anchor:middle;fill:#0000ff;stroke-width:0.264583"
           x="36.953468"
           y="11.796615">North America</tspan>
```

It's caused by the `text-anchor:middle` style of `<tspan>`. You can change the anchor of text on your SVG editor to solve this problem temporarily.

sumibin, 2023-02-22:
[image]

fixed in v4.6.2

closed 2023-02-22 by sumibin

maik-h, 2024-04-28:
Hallo zusammen ich habe das gleiche Problem bei der Version 4.11.0
Gibt es dort schon eine lösung?
[image]
[image]