Snapmaker/Luban #1878: Bug: Slicer does not work in 3d printing for 4.5.0 releated to Issue #1827

addiedavis, 2023-01-13:
# 🐞 bug report

### Affected Version(s)
The issue is caused by Luban version 4.5.0 release as of 1/12/2023

### Is this a regression? 

Yes, the previous version (4.4.0) the bug was not present. 

### To Reproduce
Steps to reproduce the behavior:
1. Open Luban
2. Click on '3D Printing'
4. Load in any .stl file
5. generate gcode
6. export gcode to workspace
7. connect to device (Snapmaker 2.0 A350)
     note: device does go to home and enclosure is able to be turned on and off, so device is fully connected
8. either select 'Start on Luban' or 'Send to Device'
     a. If 'Send to Device' was selected, then start the file on the Snapmaker device. 

Expected behavior
Snapmacker goes through the startup process -- mine were left on default so I expect the bead to move back to the printing position and the nozzle to lower to the bed. Also for both the nozzle and heated bed to start heating up. 
And for the %complete to start at 1% when it starts to move parts and prep for print. 

Current behavior
when gcode is started on Luban or on the Snapmaker no action is taken by the Snapmaker. 
The bed does not move or heat, neither does the nozzle, and the % complete remains at 0% 
Even waiting for over 10minutes, nothing changes. 

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 10 Home
[image]

- Printer: Snapmaker 2.0 A350 with 3D printer module installed

Screenshot of what it looks like on Luban after selecting "Start on Luban" for the below CaliDragon benchy:
[image]

Generated Gcode for CaliDragon benchy:
[CaliDragonv1_luban4.5.0version.zip](https://github.com/Snapmaker/Luban/files/10413267/CaliDragonv1_luban4.5.0version.zip)

parachvte, 2023-01-17:
In your G-code

```
T0
M140 S70
M105
M190 S70
```

It's waiting for the bed to heat up to 70°C.

In your screenshot, it's still 21°C. After 10 mins of heating, Is the temperature still around 21°C?

addiedavis, 2023-01-17:
Correct, the temperature stays at about 21 degrees.
With gcode generated from 4.5.0 the bed does not heat up, nor does the nozzle move to the starting position.
With gcode generated from previous versions the machine functions normally. 

[image]

addiedavis, 2023-02-03:
Ok -- I'm looking for updates since it looks like there was code merged for this issue and I'm seeing an error.

Is there a way for me to manually change the gcode so I can use my 3D printer? 
Or to get the updated luban version where this issue was fixed?

[image]

parachvte, 2023-02-04:
Just download the latest version in https://github.com/Snapmaker/Luban/releases/tag/v4.5.1

I think there is nothing wrong with your G-code. It does what it told to, the problem here is that it didn't seems to heat the bed.

May we check if the G-code is accepted by the printer, and get executed?

* If there is mismatch of firmware? Try upgrade the firmware, reset firmware settings.
* Find the differences of G-code generated by the old software (v4.4.0) and the new.

addiedavis, 2023-02-05:
Thank you @parachvte, it appears it was a firmware update needed.

closed 2023-02-05 by addiedavis