Snapmaker/Luban #1895: Bug: xxxx in 3d printing

befau, 2023-01-18:
# 🐞 bug report

### Affected Version(s)
4.5.0 and one older

Never happened before the one before 4.5.0

### To Reproduce
Steps to reproduce the behavior:
1. Open a large STL-File - starts freeze at "Loading Modell 74,0%" - not possible to stop or close Luban.
2. Log off Windows
3. Log on Windows
4. Open an other STL. Freeze at "Loading Modell 74,1%"
5. Now it is not Possible to load any STL-file at 3D-Viewer or anyone else.

It is not possible to open a STL-file.
Not in Luban
Not in 3D-Viewer in Windows

This happened now at the 2nd Windows-Laptop. I can't use them for Luban anymore. With other users on these Laptops it happened same.

Exception or Error (optional)
Starting with cmd "Snapmaker Luban.exe -v"
it shows many ok but then...

2023-01-18T20:55:53.165Z - info server:DataStorage Copy files from C:\Users\A045918E\AppData\Roaming\snapmaker-luban\Config\printing\snapmaker_j1 to C:\Users\A045918E\AppData\Roaming\snapmaker-luban\snapmaker-recover\Config-active\printing\snapmaker_j1
[14748:0118/215553.333:ERROR:entry_impl.cc(999)] Failed to save user data
[14748:0118/215553.334:ERROR:entry_impl.cc(999)] Failed to save user data
[14748:0118/215553.371:ERROR:cache_util_win.cc(20)] Unable to move the cache: Zugriff verweigert (0x5)
[14748:0118/215553.392:ERROR:cache_util.cc(146)] Unable to move cache folder C:\Users\A045918E\AppData\Roaming\snapmaker-luban\Cache\Cache_Data to C:\Users\A045918E\AppData\Roaming\snapmaker-luban\Cache\old_Cache_Data_000
[14748:0118/215553.417:ERROR:backend_impl.cc(2032)] Corrupt Index file
2023-01-18T20:55:55.806Z - warn service:schedule-task Starting ...........^v^
2023-01-18T20:55:55.837Z - warn service:schedule-task started...........^v^
Checking for update
Error: HttpError: 404
"method: GET url: https://github.com/Snapmaker/Luban/https://github.com/Snapmaker/Luban/releases.atom\n\nPlease double check that your authentication token is correct. Due to security reasons actual status maybe not reported, but 404.\n"
Headers: {
  "server": "GitHub.com",
  "date": "Wed, 18 Jan 2023 20:55:56 GMT",
  "content-type": "text/plain; charset=utf-8",
  "vary": "X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, Accept-Encoding, Accept, X-Requested-With",
  "cache-control": "no-cache",
  "strict-transport-security": "max-age=31536000; includeSubdomains; preload",
  "x-frame-options": "deny",
  "x-content-type-options": "nosniff",
  "x-xss-protection": "0",
  "referrer-policy": "origin-when-cross-origin, strict-origin-when-cross-origin",
  "content-security-policy": "default-src 'none'; base-uri 'self'; connect-src 'self'; form-action 'self'; img-src 'self' data:; script-src 'self'; style-src 'unsafe-inline'",
  "content-encoding": "gzip",
  "set-cookie": [
    "_gh_sess=R6Wab3CK6rrhKEz90XbVOo9btSrImNOdTucVREU%2FWs6H8PKX%2FAOaVjvHX%2BgK0m5nCJsLhmsi45wIpVZOOAGxcfS9RMr05bfo2ICR%2BIGjrmkLzXEF8HAkgZw3xWukl9BiGseL%2FcXb3EzS87zWAXr2ltVzFqII3RGbRDEtAaNItbm3Vd4UIQDh5ulDElLcEwrwhh1iGgmvjlfx%2BLV%2BbTEygC1H5naNn%2FtNQRiCM0QvA5kE8bmuTI08kXcMzwBLataGKgx7kFGaNbTLdOjsI%2FPQBA%3D%3D--5LWAJl2TEA5fgOP7--RflBM2bKbD6fzmyRecfLLg%3D%3D; Path=/; HttpOnly; Secure; SameSite=Lax",
    "_octo=GH1.1.1624579120.1674075356; Path=/; Domain=github.com; Expires=Thu, 18 Jan 2024 20:55:56 GMT; Secure; SameSite=Lax",
    "logged_in=no; Path=/; Domain=github.com; Expires=Thu, 18 Jan 2024 20:55:56 GMT; HttpOnly; Secure; SameSite=Lax"
  ],
  "content-length": "40",
  "x-github-request-id": "7696:BB61:45D022D:4822609:63C85CDC"
}
    at createHttpError (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:14:12)
    at ElectronHttpExecutor.handleResponse (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:106:20)
    at ClientRequest.<anonymous> (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:73:26)
    at ClientRequest.emit (node:events:390:28)
    at SimpleURLLoaderWrapper.<anonymous> (node:electron/js2c/browser_init:101:6816)
    at SimpleURLLoaderWrapper.emit (node:events:390:28)
Check for update failed HttpError: 404
"method: GET url: https://github.com/Snapmaker/Luban/https://github.com/Snapmaker/Luban/releases.atom\n\nPlease double check that your authentication token is correct. Due to security reasons actual status maybe not reported, but 404.\n"
Headers: {
  "server": "GitHub.com",
  "date": "Wed, 18 Jan 2023 20:55:56 GMT",
  "content-type": "text/plain; charset=utf-8",
  "vary": "X-PJAX, X-PJAX-Container, Turbo-Visit, Turbo-Frame, Accept-Encoding, Accept, X-Requested-With",
  "cache-control": "no-cache",
  "strict-transport-security": "max-age=31536000; includeSubdomains; preload",
  "x-frame-options": "deny",
  "x-content-type-options": "nosniff",
  "x-xss-protection": "0",
  "referrer-policy": "origin-when-cross-origin, strict-origin-when-cross-origin",
  "content-security-policy": "default-src 'none'; base-uri 'self'; connect-src 'self'; form-action 'self'; img-src 'self' data:; script-src 'self'; style-src 'unsafe-inline'",
  "content-encoding": "gzip",
  "set-cookie": [
    "_gh_sess=R6Wab3CK6rrhKEz90XbVOo9btSrImNOdTucVREU%2FWs6H8PKX%2FAOaVjvHX%2BgK0m5nCJsLhmsi45wIpVZOOAGxcfS9RMr05bfo2ICR%2BIGjrmkLzXEF8HAkgZw3xWukl9BiGseL%2FcXb3EzS87zWAXr2ltVzFqII3RGbRDEtAaNItbm3Vd4UIQDh5ulDElLcEwrwhh1iGgmvjlfx%2BLV%2BbTEygC1H5naNn%2FtNQRiCM0QvA5kE8bmuTI08kXcMzwBLataGKgx7kFGaNbTLdOjsI%2FPQBA%3D%3D--5LWAJl2TEA5fgOP7--RflBM2bKbD6fzmyRecfLLg%3D%3D; Path=/; HttpOnly; Secure; SameSite=Lax",
    "_octo=GH1.1.1624579120.1674075356; Path=/; Domain=github.com; Expires=Thu, 18 Jan 2024 20:55:56 GMT; Secure; SameSite=Lax",
    "logged_in=no; Path=/; Domain=github.com; Expires=Thu, 18 Jan 2024 20:55:56 GMT; HttpOnly; Secure; SameSite=Lax"
  ],
  "content-length": "40",
  "x-github-request-id": "7696:BB61:45D022D:4822609:63C85CDC"
}
    at createHttpError (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:14:12)
    at ElectronHttpExecutor.handleResponse (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:106:20)
    at ClientRequest.<anonymous> (C:\Program Files\Snapmaker Luban\resources\app\node_modules\builder-util-runtime\out\httpExecutor.js:73:26)
    at ClientRequest.emit (node:events:390:28)
    at SimpleURLLoaderWrapper.<anonymous> (node:electron/js2c/browser_init:101:6816)
    at SimpleURLLoaderWrapper.emit (node:events:390:28) {
  statusCode: 404,
  description: 'method: GET url: https://github.com/Snapmaker/Luban/https://github.com/Snapmaker/Luban/releases.atom\n' +
    '\n' +
    'Please double check that your authentication token is correct. Due to security reasons actual status maybe not reported, but 404.\n',
  code: 'HTTP_ERROR_404'
}

... and then after continuing loading Programm ...

2023-01-18T20:58:36.923Z - info server:DataStorage Removing files:
2023-01-18T20:58:40.550Z - error service:definition Failed to read JSON file: C:\Users\A045918E\AppData\Roaming\snapmaker-luban\Config\printing\original_single\quality.fast_print.def.json
2023-01-18T20:58:40.563Z - error service:definition Error: ENOENT: no such file or directory, open 'C:\Users\A045918E\AppData\Roaming\snapmaker-luban\Config\printing\original_single\quality.fast_print.def.json'
    at Object.openSync (node:fs:585:3)
    at Object.func [as openSync] (node:electron/js2c/asar_bundle:5:1812)
    at Object.readFileSync (node:fs:453:35)
    at Object.e.readFileSync (node:electron/js2c/asar_bundle:5:9160)
    at t.value (C:\Program Files\Snapmaker Luban\resources\app\src\server\index.js:2:698841)
    at Wh (C:\Program Files\Snapmaker Luban\resources\app\src\server\index.js:2:704687)
    at Zh (C:\Program Files\Snapmaker Luban\resources\app\src\server\index.js:2:704738)
    at C:\Program Files\Snapmaker Luban\resources\app\src\server\index.js:2:728631
    at yp (C:\Program Files\Snapmaker Luban\resources\app\src\server\index.js:2:728643)
    at Layer.handle [as handle_request] (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\layer.js:95:5)
    at next (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\route.js:144:13)
    at Route.dispatch (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\route.js:114:3)
    at Layer.handle [as handle_request] (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\layer.js:95:5)
    at C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\index.js:284:15
    at param (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\index.js:365:14)
    at param (C:\Program Files\Snapmaker Luban\resources\app\node_modules\express\lib\router\index.js:376:14) {
  errno: -4058,
  syscall: 'open',
  code: 'ENOENT',
  path: 'C:\\Users\\A045918E\\AppData\\Roaming\\snapmaker-luban\\Config\\printing\\original_single\\quality.fast_print.def.json'
}

What can i do?

xbenkozx, 2023-02-20:
I had the same issue and this is how I fixed it. 

Uninstall Luban. 
Search for Luban folder in appdata and delete. 
Reinstall Luban

RudyOnRails, 2023-03-05:
> I had the same issue and this is how I fixed it.
> 
> Uninstall Luban. Search for Luban folder in appdata and delete. Reinstall Luban

Thank you so much @xbenkozx ! I got an A350 as a gift and just got around to setting it up. I couldn't do anything with it until just now after applying your solution. Thank you so much for contributing!

bitzblitz, 2023-10-21:
For me, the problem was a missing file: 
2023-10-21T13:54:00.171Z - error service:definition Failed to read JSON file: C:\Users\bitzb\AppData\Roaming\snapmaker-luban\Config\printing\snapmaker.def.json
2023-10-21T13:54:00.172Z - error service:definition Error: ENOENT: no such file or directory, open 'C:\Users\bitzb\AppData\Roaming\snapmaker-luban\Config\printing\snapmaker.def.json'

I was able to copy over the file from an older installation on another computer and that fixed it.