Snapmaker/Luban #1903: Bug: 4.5.1 Artisan, right extruder does not show all print settings

Th199403, 2023-01-25:
Luban 4.5.1

Please add the same settings to the right extruder as the left one. as can be seen in the pictures, the right extruder has fewer adjustment options.

- Operating System: [Windows 10 x64, macOS Ventura
- Printer: [Snapmaker Artisan]

[image]
[image]

JiriHlavacek, 2023-01-27:
Forget about dual printing in Luban. I found, that many things are linked (synchronized) from left to right etruder, so you cannot setup different sizes/line widths for each extruder.

It seems like currently only some parameters are individually settable for each extruder. It seems it is currently prepared just for Snapmaker J1 with nozzles 0.4 on both extruders. It will take some time tu fully support independent settings for each on SM2 Axx0 series.

parachvte, 2023-01-30:
@JiriHlavacek Thanks for the reply, you've done some digging :)

There are actual 3 stacks of settings: the global, left extruder, right extruder. We display global settings on left extruder (current version), so you will see parameter values synchronized from left to right. We are thinking about how to reflect the 3 stacks to users without adding more complexity to the software.

Parameters that are configured by `snapmaker_modify_0.def.json`, which is an overwrite of default settings. In this file we control how parameters are organized and related (value calculated topologically), the adopt to our changes to LunarSlicer. We will do some work to tidy our *.def.json files afterwards.

If you can run the software from source code, you can change this [function](https://github.com/Snapmaker/Luban/blob/main/src/app/constants/preset.ts#L40) to control parameters to be displayed on right extruder.

Th199403, 2023-02-06:
@parachvte how about an option selection whether the settings should be mirrored or not.  then every user can decide for themselves