Snapmaker/Luban #1907: Bug: Handset Freezes During Startup

Spinoptics, 2023-02-01:
I can't get to any menus  on startup. Started after  update to 1.1.5. I tried to reset via Luban m502&m500 on serial USB connection. Still no fix. What can I do.

# 🐞 bug report

### Affected Version(s)

 The issue is caused by version x.y.z

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Windows 10 x64, macOS Big Sur, Ubuntu 20.04, etc]
- Printer: [Snapmaker Original, Snapmaker A150, etc]

</code></pre>

parachvte, 2023-02-01:
Are you mentioning the firmware?

parachvte, 2023-02-03:
Guess you are encountering a known issue, below is the solution.
--
If your screen continues to go black, this turns into another problem.

To resolve the problem, you'll need the Controller Firmware and the Firmware Update Tool. The Controller firmware is different from the firmware files you can find on our website.

You can get the Controller firmware and Firmware Update Tool from the following link: [https://drive.google.com/drive/folders/19yXYxA8X_SKqhjzTdH5bAu3fFcVe3o4L?usp=sharing.](https://drive.google.com/drive/folders/19yXYxA8X_SKqhjzTdH5bAu3fFcVe3o4L?usp=sharing)
If the PC is a MacOS Firmware Update Tool download it here: https://drive.google.com/file/d/1f2hJgDidGmEVc1YjHwNsKsxH1-rJPUMd/view?usp=sharing

Please follow the steps below to have a try if you get the firmware and tool ready.
1. Open the **Firmware Update Tool** on your computer.
2. Connect the printer to your computer **via the USB cable**.
3. Click on the **Start Update** button to load the Controller firmware file.
4. **Power on** the Power Supply.
5. Expand the dropbox, select the **correct COM port**, and click on the **Open** button immediately after powering on the printer. **Please do this quickly as there is only 7s after power on that you can flash the firmware.**
6. Wait for the firmware to be updated.

Then check if the Controller will work properly. You may refer to the attached video.

Finally, **Update** the latest firmware twice(This is important, otherwise the problem will reappear). Subsequent connection module will then request an upgrade, at this time can be upgraded normally and will not cause problems.

Latest Version: [Snapmaker2_V1.15.12](https://s3-us-west-2.amazonaws.com/snapmaker.com/download/firmware/Snapmaker2_V1.15.12.bin)