Snapmaker/Luban #1911: Bug: Thin walls in 3d printing - Sometimes generated

tyeth, 2023-02-02:
Look at this curved section with a hole close to the wall, some of the layers generate a complete wall and some do not. The model (STL) includes a vertical section there so luban should do the same for all layers and complete the wall.
[image]
[image]
[fan_duct_left.zip](https://github.com/Snapmaker/Luban/files/10573512/fan_duct_left.zip)

Luban 4.5.0 windows11

tyeth, 2023-02-02:
[SOG_PETG_Plate_6b310ee0.zip](https://github.com/Snapmaker/Luban/files/10573562/SOG_PETG_Plate_6b310ee0.zip)
Full project and all STLs, the one stl clearly showing the problem is fan_duct_left which I separately included in previous post.

parachvte, 2023-02-03:
Does it print like this? Or just a start/end of a layer (z seam)?

tyeth, 2023-02-03:
I wont waste the plastic sorry.

JiriHlavacek, 2023-02-06:
You can set little bit more or less for line width, than nozzle size diameter. For 0.4 mm it is ok between 0.35 to 0.48 mm (but with specific nozzle types you can go up to 200% of specification - need to have big outside diameter at end), personally I do not go to less than 0.3 mm for nozzle with 0.4 mm specification. So I think, it will not help you in this case. 

Thin walls (less than minimal line width) are ok for small segments (ideal for inner walls), but not for large paths. Try to buy smaller nozzle, if it is really needed.

Another possibility is experiment with seam position and wall settings.
Also look at print settings in section Structure fine tunning. There are many choices for changing behavoiur in such cases.
Look at things like Minimal even/odd/thin wall line width and Wall transition. There is good guide with pictures to understand what is what.

tyeth, 2023-02-17:
I think the model is flawed, prusa slicer has a similar set of gcode decisions. Thanks for your time as always

closed 2023-02-17 by tyeth