Snapmaker/Luban #1914: Bug: Artisan - Laser Mode - Preview Image turned around

mmack, 2023-02-04:
If i do a Camera Capture in Luban 4.5.1 with the Artisan, the Image is upside Down. Look at my example Image, the Iphone is placed with the bottom towards the door. 

[image]

parachvte, 2023-02-04:
You might need a calibration for the camera. The calibration points affect the preview.

mmack, 2023-02-04:
I did all the calibrations the Artisan forces me to do. Including the camera calibration.

parachvte, 2023-02-04:
Sorry for inconvenience.

The upside down capture image could be software problem, or hardware problem. And since we don't have a way to export the camera calibration configs we might need to recheck this can be re-produced on our side.
I would suggest you contact support@snapmaker.com for direct technical support.

closed 2023-02-21 by sumibin