Snapmaker/Luban #1919: Feature Request: J1 & Artisan & SM2.0 Dual extruder controlling function

sumibin, 2023-02-07:
## Description

Support for dual extruder control in workspace

- Idle
[image]

- Printing
[image]

## Linked Issues
[Bug: 3D Printing UI – "Load" and "Unload" buttons no longer appear in 4.5.1 #1898](https://github.com/Snapmaker/Luban/issues/1898
)

parachvte, 2023-05-19:
## Snapmaker J1

- [ ] Console, send G-code not working
- [ ] Switch Extruder
- [x] Extruder Temperature (both left and right)

dandyfopper, 2023-06-21:
waiting for this as it makes calibrating the dual extruder harder