Snapmaker/Luban #1930: Feature Request: SM2.0 bed heating finish delay (Alpha)

parachvte, 2023-02-10:
The heated bed will deform during the heating process, and waiting a certain amount of time after reaching the target temperature will make heated bed closer to the cold bed state.

[image]

Therefore, defining different bed heating finish delays for different print areas can improve the first layer adhesion.

- ≥ 300*300, 10 min
- ≥ 250*250,  8 min
- ≥ 200*200,  5 min
- ≥ 150*150,  2 min
- < 150*150, 0 min

sumibin, 2023-02-13:
[snapmaker_modify_0.def.zip](https://github.com/Snapmaker/Luban/files/10720445/snapmaker_modify_0.def.zip)

parachvte, 2023-02-13:
https://github.com/parachvte/Luban/releases/tag/v4.6.0-exp-bed-deform.1

closed 2023-02-21 by sumibin