Snapmaker/Luban #1956: Regression on adjusting temp in Luban

atherr, 2023-02-17:
Upgraded to version Luban 4.5.1 to enable the dual-extruder.  When I did, the ability to change the temp for any item (left extruder, right extruder or heated bed) is no longer available.
I can adjust the Work Speed, but all others cannot be modified.
I understand that the standby temp and the active temp are in flux, but if I am managing the print from the Luban console and see that I should go up/down 5 degrees in active if the print is not looking like it should be, or down 5 degrees in standby because the filament is dripping, I should not need to cancel the print to start over.  On my first print with this new software and the dual extruder, the nozzle clogged because the temp was too low, but I could not change it and the second filament only started after multiple layers.

TLDR: Please put back the ability to adjust the temperatures for the nozzle(s) and the heated bed

parachvte, 2023-02-18:
The temperature is taken from material.

Do you mean you want to adjust the temperature on left/right extruder, besides the seletion of material?

atherr, 2023-02-18:
On the left side of the Luban console, I could tell the 3d printer to set
the current temp to 190, for example and then go over to the printer which
is in a different room of my house, and change the filament.  This compares
to waiting for the extruder to heat up when setting the temp on the
handheld console attached to the printer.

Also, if I see the settings are wrong (say I am using a filament from a
vendor for the first time) and would like to adjust it while printing from
Luban, I could previously change those settings while printing was
occurring.  Instead of just displaying the current temperature of the
extruder on the Luban console, I could also change it.

As I mentioned in the PR, I understand that the temp is changing based on
which extruder is active in the dual extruder module, but the change is a
regression as changes to the temp was allowed previously and now it is not.

parachvte, 2023-02-21:
> On the left side of the Luban console, I could tell the 3d printer to set the current temp to 190, for example and then go over to the printer which is in a different room of my house, and change [...]

Sorry I got you wrong. I agree adjustments on temperatures for dual extruder is a strong point.