Snapmaker/Luban #1987: BUG: Synchronize nozzle diameter will change wrong nozzle size

sumibin, 2023-02-22:
## Versions
v4.6.1

## Steps

1. Change Nozzle Diameter to 0.6 on both Nozzles
2. Start Workspace
3. Connect to the printer (Artisan 0.4+0.4) via wifi
4. pop-up prompt, while changing the left nozzle diameter to 0.4

## Correct
1. pop-up prompt
2. Yes -> Change both nozzle diametet
3. No -> No change

matousrokos, 2023-03-05:
Hi, I get endless loop of confirmation requirement when changing hot end 
[image]

Confirming doesn't change anything. Where in Luban I can set new nozzle size? I've even tried config files, yet no difference.

Bockhorn-IT, 2023-03-10:
Same error here, endless loop 
[image]