Snapmaker/Luban #2008: Bug: Luban reports error on checking for updates

colombod, 2023-02-28:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.6.3

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'Help'
2. Click on 'Software Update'
3. See error
[image]

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 10 x64
- Printer: Snapmaker A350

</code></pre>

Andygol, 2023-03-01:
I confirm the same issue for Luban 4.6.3 on macOS, M1

tensile23, 2023-04-04:
This issue has been fixed in v4.7.2

closed 2023-04-04 by tensile23