Snapmaker/Luban #2015: Bug: Luban crashes when creating toolpath for laser

Walshe-d, 2023-03-03:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.6.3

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was definitely 4.3.2, I'm not sure if versions between 4.3.2 and 4.6.2. had this issue

### To Reproduce
Steps to reproduce the behavior:
1. Go to luban, laser, 3 axis, insert text. The word snapmaker will appear.
2. Click on 'next'
3. Click on the text and then click on create tool path
4. The app crashes with no error code
 Luban crashes when I try to create a tool path for text.
[image]

Exception or Error (optional)
<pre><code>

There is no error code
</code></pre>

Expected behavior

I expect the toolpath to be created

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: [Ubuntu 20.04.5 LTS]
- Printer: [Snapmaker Original]

</code></pre>

Walshe-d, 2023-03-03:
When luban prompts to specify what printer you are working with if you choose snapmaker original **with** module extension this error will occur. 

However, if you use the snapmaker original **without** module extension, this error does not occur.

This actually probably wasn't an issue with v4.6.3 apologies

parachvte, 2023-03-17:
Bug confirmed on v4.6.3, Ubuntu 20.04 LTS.

It should be fixed in v4.7.0 now. I'm closing the issue.

Watson-Sei, 2023-10-17:
No, the problem still occurs.
It occurs in Snapmaker Luban 4.9.1 when creating toolpaths for CNC 3-axis.
[image]

Watson-Sei, 2023-10-17:
@parachvte 
No, the problem still occurs.
It occurs in Snapmaker Luban 4.9.1 when creating toolpaths for CNC 3-axis.
[image]

ExtraZiadeh, 2023-10-21:
@parachvte 

I can confirm with @Watson-Sei that this is still an issue with Luban 4.9.1