Snapmaker/Luban #2021: Machinist fonts on CNC

Moritz8149, 2023-03-09:
Hi,

Sadly i have a Problem with importing single line machinist fonts into the luban program.
i have a set of 17 single line fonts, but none of them works properly in luban.
i can import them, aswell as select them, but the shown text is just a few circles or sometimes completly blank, depending on the used font.

is there any way around this problem without using inkscape?

best regards 
Moritz
[image]

tensile23, 2023-05-09:
I tried some single-line fonts, but they all render properly. Can you share this font? 

[image]

Moritz8149, 2023-05-09:
> I tried some single-line fonts, but they all render properly. Can you share this font? 
> 
> [image]
> 
> 

Unfortunately i cant share the Fonts as its a licensed and paid Font. 

But the Fonts are arial Styled Fonts, Just single Line for the cnc