Snapmaker/Luban #2024: Bug: Wrong scaling when import dxf files

Obicom, 2023-03-11:
**Bug report**
If I import a dxf file in Luban 4.6.3 it is always with the wrong scaling.
It is shown as a very small object and I have to resize it manualy.
If I open the file in Fusion360 or Inkscape the scaling is fine.

**Expected behavior**
If I import a dxf file in Luban I would expect the right scaling of the objects.

**My Environment**
Snapmaker Artisan with latest firmware
Luban 4.6.3 (Windows version)

Voodoofly, 2023-12-11:
i have the same problem, i export from Fusion360  the DXF File, when import it in Luban, its looks so 10x smaller, this bug hold on over 3 Years now !!!!!   Looks they cant fix it, why ever

afresh1, 2023-12-18:
I was able to reproduce this with a very simple square exported from OpenSCAD.  It seems broken both an SVG and DXF.  The OpenSCAD script saw just `square(100);` and then I exported as SVG and DXF and imported into a 3-axis laser job where they showed up as 10mm.  

[square.zip](https://github.com/Snapmaker/Luban/files/13699290/square.zip)

afresh1, 2023-12-18:
I forgot to mention that this was with Luban 4.10.1 on Windows. 

The reason I believe this should work is when I generate an stl from an OpenSCAD file with `cube(100);` in it and import it into a 3d-print project, it imports as a 100mm on a side cube and is what I expect.

[cube.zip](https://github.com/Snapmaker/Luban/files/13707869/cube.zip)

lyrradttenra, 2023-12-25:
i have the same issue with ver 4.10.1 in laser or cnc scaling is way out had to goto ver 4.9.0 worksfine pkease fix latest ver

Lumino10, 2024-01-14:
I have the same problem. When I import a simple rectangle, the mass times 10 works. I have a project with connectors. The general dimensions are correct, but the connectors do not line up. The dimensions in Fusion 360 are correct. Also when I load it into Adobe Illustrator the connectors line up. I think this is a rounding issue as Luban only supports one decimal place. The only possibility is to put all my parts on one DXF. Unfortunately this makes Luban almost useless for laser cutting. I used Luban 4.10.1.