Snapmaker/Luban #2070: Bug: Console disappears when I connect to printer

iloving, 2023-03-24:
# 🐞 bug report

I am trying to use the console to tweak the settings on the printer, but the console is only available until I connect to the printer!

### Affected Version(s)
4.7.0

### Is this a regression? (optional)

Unknown

### To Reproduce
Steps to reproduce the behavior:
1. Go to Workspace
2. Connect to printer, either by wifi or serial
3. Scroll down to '....'
4. See error
 A clear and concise description of the problem...

Exception or Error (optional)
N/A

Expected behavior
I expect the console to _stay there_ when I connect.  As it is, the console is completely useless.

## 🌍  Your Environment

**Platform:**

- Operating System: MacOS 13.2.1
- Printer: Artisan

ijspear, 2023-04-01:
This is great news, thanks for working on this! Setting up CNC jobs without the console is very tricky.

ijspear, 2023-04-21:
@tensile23 Is there any update on this? I’m dying to get this feature back.

parachvte, 2023-04-21:
> @tensile23 Is there any update on this? I’m dying to get this feature back.

With latest software release, you have the Console back for Artisan.

We are bring the Console back to J1 next release.