Snapmaker/Luban #2099: Bug: [Development] Cannot find module '@snapmaker/luban-platform'

blakeglucas, 2023-03-31:
# 🐞 bug report

### Affected Version(s)
Luban v4.7.1

### Is this a regression? (optional)

Unknown

### To Reproduce
Steps to reproduce the behavior:
1. On a fresh clone of the Snapmaker Luban repository (preferably in an isolated environment like VirtualBox), attempt to perform the instructions under [Development](https://github.com/Snapmaker/Luban/blob/main/docs/Development.md). I've successfully reproduced this problem on Windows, Ubuntu WSL, and Ubuntu VirtualBox.
2. Using NVM, install Node 12 (former LTS) per instructions and activate it (`nvm use 12.xx.yy`).
3. Run `npm install`
4. Run `npm run dev`, again per instructions.

Exception or Error (optional)
<pre><code>src/app/constants/machines.ts:1:64 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine, MachineToolHeadOptions, ToolHead } from '@snapmaker/luban-platform';
                                                                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-2-toolheads.ts:1:31 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { ToolHead } from '@snapmaker/luban-platform';
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-2-toolheads.ts:2:30 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { ToolHeadType } from '@snapmaker/luban-platform';
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-a150.ts:1:30 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine } from '@snapmaker/luban-platform';
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-a150.ts:2:29 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { MachineType } from '@snapmaker/luban-platform';
                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-a250.ts:1:38 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { Machine, MachineType } from '@snapmaker/luban-platform';
                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-a350.ts:1:38 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { Machine, MachineType } from '@snapmaker/luban-platform';
                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-artisan.ts:1:31 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine, } from '@snapmaker/luban-platform';
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-artisan.ts:2:29 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { MachineType } from '@snapmaker/luban-platform';
                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-j1.ts:1:40 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine, ToolHead } from '@snapmaker/luban-platform';
                                         ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-j1.ts:2:54 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { MachineType, ToolHeadType, PrintMode } from '@snapmaker/luban-platform';
                                                       ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original-extended.ts:1:30 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine } from '@snapmaker/luban-platform';
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original-extended.ts:2:29 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { MachineType } from '@snapmaker/luban-platform';
                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original-toolheads.ts:1:31 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { ToolHead } from '@snapmaker/luban-platform';
                                ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original-toolheads.ts:2:30 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { ToolHeadType } from '@snapmaker/luban-platform';
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original.ts:1:30 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import type { Machine } from '@snapmaker/luban-platform';
                               ~~~~~~~~~~~~~~~~~~~~~~~~~~~

src/app/machines/snapmaker-original.ts:2:29 - error TS2307: Cannot find module '@snapmaker/luban-platform' or its corresponding type declarations.

import { MachineType } from '@snapmaker/luban-platform';
                              ~~~~~~~~~~~~~~~~~~~~~~~~~~~
</code></pre>

Expected behavior
According to the Development instructions, I'd expect a React Electron application to open. Instead it throws these errors and then continues building. The same error occurs after issuing `rs` command to nodemon or just running `npm run start-dev`.

I've tried adding `-r tsconfig-paths/register` to the gulp exec command, copying packages to the `output/` directory, but nothing helps. I'd appreciate some updated developer documentation.

## 🌍  Your Environment

**Platform:**
<pre><code>- Operating System: Windows 11 x64, Ubuntu 22.04 WSL x64, Ubuntu 22.04 x64 (VirtualBox)
- Printer: Snapmaker A350T (unrelated)
- Luban v4.7.1
</code></pre>

parachvte, 2023-03-31:
Sorry for outdated documents.

Upgrade your Node.js to 16+, everything should be fine.

blakeglucas, 2023-03-31:
Thanks for the quick response, upgrading Node to 16 worked on the Ubuntu systems. Unfortunately I'm still running in to build issues on Windows, specifically with the `build/prebuild-dev.sh` script:

```
> npm run prebuild-dev

> snapmaker-luban@4.7.1 prebuild-dev
> bash build/prebuild-dev.sh

build/prebuild-dev.sh: line 3: node: command not found

> snapmaker-luban@4.7.1 pkgsync
> node build/pkgsync.js

/mnt/c/.../Luban/src /mnt/c/.../Luban
/mnt/c/.../Luban/node_modules/.bin/cross-env: 11: exec: node: Permission denied
/mnt/c/.../Luban/node_modules/.bin/cross-env: 11: exec: node: Permission denied
/mnt/c/.../Luban
Copying print settings...
Copying other resources..
```

The `/mnt/c/` lines seem to indicate it's running WSL to match the shebang directive /bin/bash. Any ideas what's going on here? Could just be an issue with my system.

After running some of the prebuild commands manually, I was able to get the Windows build to work. I'll go ahead and close.

closed 2023-03-31 by blakeglucas

parachvte, 2023-03-31:
> The `/mnt/c/` lines seem to indicate it's running WSL to match the shebang directive /bin/bash. Any ideas what's going on here? Could just be an issue with my system.

We didn't use Windows for our daily development.
But yes if we do, we use WSL (v2) to run the bash script.