Snapmaker/Luban #2118: Bug: Camera Capture not working most of the times

Necktwi, 2023-04-09:
# 🐞 bug report

### Affected Version(s)
The issue is caused by version 4.7.2

### To Reproduce
1. Connect your Snapmaker 2.0 A350T over WiFi in Workspace
2. Go back to project and click Camera Capture > Add Background > NormalMode > Start
3. Only 1st of 9 images or none will shown.

### Exception or Error (optional)
```
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 Launching App snapmaker-luban@4.7.2...
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 THREE.Object3D.add: object not an instance of THREE.Object3D. undefined
add @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 THREE.Object3D.add: object not an instance of THREE.Object3D. undefined
add @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 REFACTOR Marlin:settings Object
value @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 machine connected via undefined.
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 REFACTOR move:status
value @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 REFACTOR move:status
value @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 TypeError: Cannot set property context of #<Gs> which has only a getter
    at e.value (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:5843276)
    at n.value (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:5851386)
    at ys (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3183842)
    at ks (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3185632)
    at Ll (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3197660)
    at t.unstable_runWithPriority (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3218146)
    at Wi (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3141022)
    at Al (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3195621)
    at yl (app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3191732)
    at app.171bea82b100406977ed.bundle.js?_=1680492295097:2:3141245
value @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 THREE.WebGLRenderer: Context Lost.
app.171bea82b100406977ed.bundle.js?_=1680492295097:2 DefinitionManager.init laser laser/a350_1600mw
:57672/#/laser:1 Uncaught (in promise) undefined
:57672/#/laser:1 Uncaught (in promise) undefined
:57672/#/laser:1 Uncaught (in promise) undefined
:57672/#/laser:1 Uncaught (in promise) undefined
Promise.then (async)
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
u @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
n @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
s @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Promise.then (async)
n @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
s @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Promise.then (async)
n @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
s @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
oe @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Ve @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Xe @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Or @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Mr @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
De @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
(anonymous) @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Nr @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Jt @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Qt @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
t.unstable_runWithPriority @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Wi @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Le @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
Zt @ app.171bea82b100406977ed.bundle.js?_=1680492295097:2
```

###Expected behavior
The image of work area in parts of 9 images should be stitched together and added as background.

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 10 x64
- Printer: Snapmaker A350T

tensile23, 2023-04-10:
Hi,感谢你的反馈。我们做了一个简易的连接测试，

Necktwi, 2023-04-14:
[image]

Necktwi, 2023-04-14:
[Luban.log](https://github.com/Snapmaker/Luban/files/11230162/Luban.log)

Necktwi, 2023-04-16:
@tensile23, please let me know if you need any additional data. Thanks!

GowthamKudupudi, 2023-04-16:
me too facing same issue

nguzeldere, 2023-07-24:
Same issue for me too. Camera capture work only if I turn led lghting off. We are expecting a solution for the issue asap.

Rolzad73, 2023-08-11:
I have had this issue for a long time, it would stop the capture process somewhere in the middle of taking images.

- 1st I updated both the machine (A350 orig upgraded with new rails) to firmware x.x.x and Luban to 4.8.2 and it still failed.
- I read the comments above about the LED lighting issue so decided to try with enclosure LED off and no extra light, the 1st image would not even capture (software just stopped there).
- Next I tried still with the enclosure LEDs off but added a CFL lamp inside and it worked without any issues, tested multiple times.

Perhaps, because digital cameras don't behave well sometimes with LED lighting because LED dim control is achieved by pulse flashing thus causing either lines or maybe the camera takes a picture DURING a dark "off" pulse causing some kind of freeze in the software because it sees nothing in the picture? <--speculation, but maybe worth invesigating.

jpheos, 2023-12-20:
I have the same problem here

Necktwi, 2024-06-24:
i regret buying snapmaker, i didn't use my snapmaker since i bought it 1 year back due to these bugs

lionB777, 2025-03-23:
Same problem here, it worked fine all day, now out of the blue, no matter if I recycle the printer power, restart the computer, it doesn't work anymore.