Snapmaker/Luban #2140: Connection failed. The machine has been connected to other device.

prat67, 2023-04-21:
# 🐞 bug report

### Affected Version(s)
Luban 4.7.2 & 4.7.3

### To Reproduce
Steps to reproduce the behavior:

Click Connect to a previously working device that was connected, and you get the error message
"Connection failed. The machine has been connected to other device. "

The ONLY solution is to delete the machine.json file in the APP Data User roaming folder (as suggested in forum posts)
But you have to do this every time the printer is turned off and on. 

Really annoying bug. 

## 🌍  Your Environment

**Platform:**

- Operating System: Windows 10 x64
- Printer: [Snapmaker A350]

</code></pre>

tensile23, 2023-04-23:
Please check if the firmware version of your machine is too old. 
Here is how to upgrade the firmware:
[How to Upgrade the Firmware of Snapmaker 2.0](https://support.snapmaker.com/hc/en-us/articles/360051875394-How-to-Upgrade-the-Firmware-of-Snapmaker-2-0)

closed 2023-04-23 by tensile23

prat67, 2023-04-25:
No the firmware is up to date. Why would you close this? 
It's a KNOWN issue...