Snapmaker/Luban #2141: Bug: images with transparency broken in laser project

iloving, 2023-04-25:
# 🐞 bug report
I have a PNG with a round image that I am trying to precisely place onto the material I am engraving.
However when I try to import the PNG, it loads fine for a second, and then suddenly the transparent background becomes completely white.  This blocks the background picture I took of the material and makes it impossible to position the image.

I've also tried saving the image as an SVG, but when Luban loads it, it complains that the image is larger than the work surface and then doesn't display the image at all.

This has completely blocked me from completing my project.  I have attached the images in question for review.
I have not tested whether the same issue occurs in CNC mode.

### Affected Version(s)

 The issue is caused by version x.y.z
4.73

### Is this a regression? (optional)
Unknown

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Import one of the images into a laser project.
 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

If an image has a transparent background, I expect Luban to respect that transparency when loaded.

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Mac OS 13.3.1
- Printer: Artisan
</code></pre>
[image]

[image]

iloving, 2023-04-25:
Unless I've completely misunderstood the processing modes?  I just switched to Vector mode and it's doing exactly what I needed...

Rolzad73, 2023-08-11:
I see this as well on Luban 4.8.2
Vector mode shows transparency, all other modes make the transparent part white opaque by default.

KuntzeM, 2023-11-29:
I see this in Luban 4.10.1, too. 
This leads to an inaccuracy in alignment if several images are used for engraving and cutting.