Snapmaker/Luban #2165: Little things i found in 4.8 alpha

nfmaas, 2023-05-15:
@parachvte It would be nice to have an undo button, so you can undo the last color action if you click it once too much. Now you need to start over again or choose the other colour to paint it back to that colour.
Next thing is the middle mouse button pushed to rotate the model. We can now with the right button drag but rotating would be handy as well.

parachvte, 2023-05-17:
> Next thing is the middle mouse button pushed to rotate the model. We can now with the right button drag but rotating would be handy as well.

You can rotate as normal, by dragging the scene.

> It would be nice to have an undo button

Yep. That's the next step (probably in next version), an undo action is handy when coloring.
To make current version easier, we bind keyboard 1 and 2 to quickly switch between left extruder and right extruder.

parachvte, 2023-09-07:
Both optimization mentioned above has been implemented in v4.8.1 and v4.8.2, closing this issue.

closed 2023-09-07 by parachvte