Snapmaker/Luban #2186: Bug: xxxx in 3d printing

Obicom, 2023-05-27:
# 🐞 bug report

### Luban Version 4.8.0 (and before)

### Is this a regression? (optional)

Yes

### To Reproduce
Steps to reproduce the behavior:
Tray to egrave with the 10 Watt lser and rotary module a 360° picture (PNG) on a 23mm (diameter) tube
I adjust the origin workpoint
Connect Luban via Wifi and start the job ... 
Luban ask to start from work origin - confirm with YES
all works good.

If I do exactly the same via Luban USB connection
Luban ask to start from work origin - confirm with YES
The laser head moves down and crashes into the object - WHY???

Exception or Error (optional)
With the USB connection the jobs starts not from the defined origin workpoint.
Laser head crashes into the object !!! This happens only with USB connection, not with WiFi connection.

## 🌍  Your Environment

**Platform:**
Windows 11
Luban 4.8.0 release version (latested)
Snapmaker Artisan with Rotary module and 10W laser

Obicom, 2023-07-16:
no update so far? From my point of view it is a major bug and should be solved ASAP

tensile23, 2023-07-18:
https://github.com/parachvte/Luban/releases/tag/v4.9.0-alpha.1
我们已经确认了这个问题，会在后续的正式版本中修复它。上面是一个修复的测试包。

Obicom, 2023-07-18:
@tensile23 thank you very much for the information and link. I checked it out and issue is solved. Looking forward for the next official release (v4.9.0)

closed 2023-10-07 by tensile23