Snapmaker/Luban #2187: Bug: Laser - with vertical lines on a 360° rotation 10° are missing

Obicom, 2023-05-27:
Luban Version 4.8.0 (and before)
Is this a regression? (optional)
Yes

To Reproduce:
Steps to reproduce the behavior:
Tray to egrave with the 10 Watt lser and rotary module a 360° picture (PNG) on a 23mm (diameter) tube
Picture is exactly 72,3x49mm (23mm diameter x PI = 72,3mm)
Settings: 0,1 lines , fill mode - VERTICAL lines
Connect Luban via Wifi and start the job ...
Luban ask to start from work origin - confirm with YES
Job start working but on the end 10° are missing (ca. 6.4mm are not engraved) - WHY???

If I do exactly the same via Luban and 0,1 lines , fill mode - HORISONTAL lines
Luban ask to start from work origin - confirm with YES
Job start working  and on the end 360° are engraved with exactly the same settings. Only changed from vertical lines to horizontal lines.

🌍 Your Environment
Platform:
Windows 11
Luban 4.8.0 release version (latested)
Snapmaker Artisan with Rotary module and 10W laser

tensile23, 2023-05-30:
Can you provide us with a Gcode and an image so that we can reproduce your question?

tensile23, 2023-05-31:
We tried to reproduce your problem on the machine. However, Gcode was able to work well on our machine and the processed part was only a small blank segment.

This issue should not be caused by Luban generating Gcode, please get in touch with technical support for other help (providing this issue will help technical support to troubleshoot the problem.)

Below is the contact information for technical support.
- Technical Support: support@snapmaker.com

closed 2023-05-31 by tensile23

Obicom, 2023-06-19:
@tensile23 : I had the great chance to get a second rotary module (new one) for testing.
What shall I say, exactly the same problem with the second, brand new, rotary module?
What does that mean? We can exclude the rotary module as the reason for this problem.
From my point of view it must be a software( Luban) or Firmware problem. Or what do you think about it?
I attached two new pictures from the new rotary module outcome. The first one shows the 10% gap if I do  verticale engraving on the upper site and below you can see if I do exactly the same with horizontal lines there is no gap.
All round logos are inclined to the right and normal Arial characters look like italic style. (Second picture) This is really a big problem for me.

Please advise, what can I do? In this condition I can't work with the 4-axis mode.Please help.
Thank you in advance.

Best regards, Dirk
[image]
[image]
[image]