Snapmaker/Luban #2197: Bug: Can't set Seam to midpoint of wall in 3d printing

tyeth, 2023-05-29:
V4.8.0
Model is a square with round corners that stick out past the square walls. I want the seam not around the corners, I tried every mode and option and messed with x/y and relative but never got the seam to be anywhere but one of the corners.

Allow a user to paint the seam position like support, or at least to specify the X and Y position with a mouse by clicking on the model.

Please allow one of your developers to try and use luban to get the seam halfway along one of the walls in this STL.
[pole_post_cap-BodyPad_test72mm.zip](https://github.com/Snapmaker/Luban/files/11591436/pole_post_cap-BodyPad_test72mm.zip)

tensile23, 2023-05-31:
[image]
We have determined that the engine is valid for the configuration parameters. However, the algorithm still prefers to generate seams at the four corners.

We prefer a feature request rather than a bug for painting seam locations.

tyeth, 2023-05-31:
I accept painting seams is a feature request. This ticket is a bug to state it's impossible to use the interface X and Y to move the seam as desired or expected. If it's not impossible then please provide the X and Y to move the seam to the midpoint of a long wall rather than the corners