Snapmaker/Luban #2200: Bug: xxxx in 3d printing

vdborght, 2023-05-29:
# 🐞 bug report

### Affected Version(s)
Luban 4.8.0

### To Reproduce
Steps to reproduce the behavior:
1. Import a model
2. Color parts of the model
3. Save the project
4. Close an re-open Luban
5. Open the project from before

Experienced behaviour
Coloring of the model is still present
See screenshots and snap3D project attached

Expected behavior
Coloring is kept

## 🌍  Your Environment

**Platform:**
- Operating System: macOS ventura
- Printer: Snapmaker J1

Colored models before saving the project
[image]

Loading the same project after closing and re-opening Luban
[image]

Project file
[DualColorBug.snap3dp.zip](https://github.com/Snapmaker/Luban/files/11593501/DualColorBug.snap3dp.zip)

tkw76, 2023-06-04:
I had the same thing happen as well.

parachvte, 2023-06-16:
Has been fixed in v4.8.1

closed 2023-06-28 by parachvte