Snapmaker/Luban #2202: Bug: App crashs if I want to draw  a line in laser mode

Obicom, 2023-05-30:
# 🐞 bug report

### Affected Version(s)
4.8.0

### Is this a regression? (optional)
No

### To Reproduce
Steps to reproduce the behavior:
1. Go to 'New project'
2. Click on 'Pen' to draw a line
4. See error

[image]

Expected behavior
No error message - would like to draw  a line

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11
- Printer: Snapmaker Artisn

</code></pre>

tensile23, 2023-05-31:
We will fix it in the next fix version.

ltmetalman, 2023-06-09:
I am having this problem still, when can we expect the fix? This seems to have been an issue for a week now. I am just trying to use my laser to cut simple shapes. Right now its a humongous pain to just make a triangle...

parachvte, 2023-06-12:
@ltmetalman Sorry for bug. A patch version of Luban v4.8.0 will be released this month (v4.8.1).

tyeth, 2023-06-13:
@parachvte It's treated as a non-issue to assume a customer will roll-back to a working version or not be severely impacted by show-stopper level bugs. That's usually not the case, people just start to resent not being able to use their snapmaker.
 Maybe we (customers) can assume it's a non-issue to release a quick hot-fixed version every time a new show-stopper level bug is introduced in a minor release. Fair enough to not hot-fix everything, but any major regressive bugs should have <7day fix-time

ltmetalman, 2023-06-29:
Seems to be fixed now... things are much easier now, thank you.

parachvte, 2023-06-29:
It was an easy fix, sorry for the delay of release (with another laser canvas related issue to be resolved).

closed 2023-06-29 by parachvte

ltmetalman, 2023-06-29:
> @parachvte It's treated as a non-issue to assume a customer will roll-back to a working version or not be severely impacted by show-stopper level bugs. That's usually not the case, people just start [...]

It was an easy fix. No worries 'mahn.