Snapmaker/Luban #2225: Crash on launch on macOS 14 Beta

pixelcmtd, 2023-06-11:
# 🐞 bug report

### Affected Version(s)

4.8.0

### To Reproduce

```sh
"/Applications/Snapmaker Luban.app/Contents/MacOS/Snapmaker Luban"
```

```
2023-06-11 13:09:36.905 Snapmaker Luban[6331:244074] WARNING: Secure coding is not enabled for restorable state! Enable secure coding by implementing NSApplicationDelegate.applicationSupportsSecureRestorableState: and returning YES.
err ERR_INVALID_URL (-300) loading ''
[6331:0611/130936.977660:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.012130:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.047992:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.062093:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130937.084349:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.118963:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.152764:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.187174:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.223184:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.240088:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130937.259107:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.292722:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.325498:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.359837:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.394066:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.407647:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130937.428861:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
2023-06-11T11:09:37.449Z - info util-fs Initialize data storage at directory: /Users/pixel/Library/Application Support/snapmaker-luban
[6331:0611/130937.464152:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.497660:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.533714:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.568880:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.588878:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130937.604167:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.637309:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
2023-06-11T11:09:37.669Z - info util-fs Loading configuration from "/Users/pixel/.snapmaker-luban.json"
2023-06-11T11:09:37.670Z - info util-fs Initializing user data storage...
[6331:0611/130937.671157:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.706807:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.740736:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
(node:6334) Warning: Accessing non-existent property 'padLevels' of module exports inside circular dependency
(Use `Snapmaker Luban Helper --trace-warnings ...` to show where the warning was created)
2023-06-11T11:09:37.749Z - info util-fs Starting the server at http://127.0.0.1:58304
Clearing folder /Users/pixel/Library/Application Support/snapmaker-luban/Tmp
Clearing folder /Users/pixel/Library/Application Support/snapmaker-luban/Sessions
(node:6331) ProtocolDeprecateCallback: The callback argument of protocol module APIs is no longer needed.
(Use `Snapmaker Luban --trace-warnings ...` to show where the warning was created)
2023-06-11T11:09:37.754Z - info util-fs Clearing folder /Users/pixel/Library/Application Support/snapmaker-luban/Tmp
2023-06-11T11:09:37.755Z - info util-fs Clearing folder /Users/pixel/Library/Application Support/snapmaker-luban/Sessions
2023-06-11T11:09:37.761Z - info util-fs Starting the server at http://192.168.178.50:58304
2023-06-11T11:09:37.761Z - info util-fs Starting the server at http://192.168.178.65:58304
[6331:0611/130937.772680:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130937.775674:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.808572:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.855942:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.894775:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
2023-06-11T11:09:37.928Z - info util-fs upgrade config files in folder /Users/pixel/Library/Application Support/snapmaker-luban/Config
2023-06-11T11:09:37.928Z - info util-fs Copying folder /Users/pixel/Library/Application Support/snapmaker-luban/Config to /Users/pixel/Library/Application Support/snapmaker-luban/snapmaker-recover/Config-active
[6331:0611/130937.932296:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.975180:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130937.980533:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130938.010512:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.055573:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.091222:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.126680:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.127154:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130938.161579:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.196262:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.241704:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.271609:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130938.279343:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.312822:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.345659:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.378548:ERROR:network_service_instance_impl.cc(975)] Network service crashed, restarting service.
[6331:0611/130938.397364:ERROR:gpu_process_host.cc(979)] GPU process exited unexpectedly: exit_code=9
[6331:0611/130938.397388:FATAL:gpu_data_manager_impl_private.cc(448)] GPU process isn't usable. Goodbye.
[1]    6331 trace trap  "/Applications/Snapmaker Luban.app/Contents/MacOS/Snapmaker Luban"
[6400:0611/130938.412142:ERROR:mach_port_rendezvous.cc(310)] bootstrap_look_up com.snapmaker.luban.MachPortRendezvousServer.1: Permission denied (1100)
[6400:0611/130938.412654:ERROR:child_thread_impl.cc(228)] Mach rendezvous failed, terminating process (parent died?)
```

## 🌍  Your Environment

**Platform:**

- Operating System: macOS Sonoma 14.0 Beta 23A5257q

mehrbodnahal, 2023-07-23:
same happens to me. I believe it is related to mac OS 14 beta

pixelcmtd, 2023-07-25:
@mehrbodnahal My bet is on Apple having removed some API (or possibly a new codesigning requirement) they deprecated like 3 years ago after introducing a replacement 5 years ago that some dependency of Luban is still using for some reason. I've had similar problems with all kinds of professional software (NLEs, VST/AU plugins and their licensing crap, ...) on every new major macOS version.