Snapmaker/Luban #2241: Bug: Presets not loading values properly

fvlasie, 2023-06-22:
# 🐞 bug report

### Affected Version(s)
Only tested in 4.8.0 Laser 3-axis

### To Reproduce
Steps to reproduce the behavior:
1. Go to Laser 3-axis Worksapce
2. Create a shape .
3. Click on Create Toolpath
4. Choose Preset: Black Anodized Aluminium—Vector Engraving
5. Work Speed will show 3500 mm/min which is too fast.
6. Choose Method Fill then choose Method On the Path
7.  Work Speed will now read 140 mm/min

Expected behavior
Work Speed should read 140 mm/min without having to toggle back and forth between methods.

## 🌍  Your Environment

A350 10w Laser Module
- Operating System: macOS Ventura 13.4
- Printer: A350 with 10w Laser Module

https://github.com/Snapmaker/Luban/assets/33334298/d4240925-ad1a-491d-a6a6-9e43500c0a73

tensile23, 2023-07-05:
The vector engraving speed for black anodised aluminium is 3000mm/min at 10W laser, which should work fine in the material we have included. We would like to see what kind of workpiece you engrave and we will consider optimising or adding new configurations.

fvlasie, 2023-07-06:
I find 3500mm/min too fast for most anodised aluminium I have tried. I get better results with about 500mm/min.

tensile23, 2023-07-06:
> I find 3500mm/min too fast for most anodised aluminium I have tried. I get better results with about 500mm/min.

In this case, you can create a custom Profile to make it easier to use. At the button pointed to in the picture below, you can access the preset manager.

We will add a preset or modify it after sufficient testing.