Snapmaker/Luban #2249: Information in Chinese only about the available update

Andygol, 2023-07-02:
# 🐞 bug report

### Affected Version(s)

 
The problem was reported on Facebook - https://www.facebook.com/groups/snapmaker/posts/1764946907256948/ and I encountered it as well.

[image]

This issue has existed since the implementation of the following code 👇 

https://github.com/Snapmaker/Luban/blob/e9e67156162d8ea66f58860b2001a5f7b01c854b/src/main.js#L138-L160

as there are no any other sources to get release notes except https://snapmaker.oss-cn-beijing.aliyuncs.com/snapmaker.com/download/luban/Snapmaker-Luban-4.8.1.changelog.md

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: ....

### To Reproduce
Steps to reproduce the behavior:
1. Go to Settings->Preference
2. Click on 'Check for updates'
    If a newer version is available, you will only receive information in Chinese, regardless of your user interface language.

[image]

 A clear and concise description of the problem...

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

It should be possible to receive information about the availability of a new release in the user's language, and in the absence of the translation, the release information should be in English

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows, macOS
- Printer: any supported

</code></pre>

tensile23, 2023-07-05:
In some cases, this exception is triggered when the github changelog is not retrieved, and we will fix it in the future.

FriedSemiconductors, 2023-08-11:
@Andygol yes the last update notifications were all in chinese for me
and others are reporting the same on facebook.

for example:
https://facebook.com/groups/snapmaker/permalink/1790407614710877/

parachvte, 2023-09-13:
This should be fixed once you are in v4.8.2.

Close this issue when v4.9 is released.