Snapmaker/Luban #2289: Bug: Enclosure "Door Detection" toggle switch missing

Rolzad73, 2023-08-11:
# 🐞 bug report

### Affected Version(s)
4.8.2

### To Reproduce
Steps to reproduce the behavior:
1. Have Snapmaker 2 with Enclosure
2. Check "Enclosure ui section" (center right of screen)
3. Notice that "LED Strips" and "Exhaust Fan" toggle switches exist, but "Door Detection" is missing

BEFORE (pre-update was Luban 4.7.3?)
[image]

AFTER (Luban 4.8.2)
[image]

Expected behavior
Door Detection toggle should be available

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 10 x64
- Printer: Snapmaker A350T

</code></pre>

Wynand-S, 2023-10-05:
I noticed the same in Luban v 4.9.1 - No option to toggle the door detection

Rolzad73, 2023-10-06:
It looks like it was removed in this commit 407dc7ef56bda4f4632637fa4d7b4c9152b6a98d
Why remove that feature?

tensile23, 2023-10-07:
On SM2 and Artsian, we removed this feature for security reasons.

closed 2023-10-07 by tensile23

Rolzad73, 2023-10-07:
> On SM2 and Artsian, we removed this feature for security reasons.

What security reason? this isn't a toy made for children.

Rolzad73, 2023-10-07:
> I noticed the same in Luban v 4.9.1 - No option to toggle the door detection

looks like the have "security" reasons for removing it. Alternatively you can remove the connection for the door detection manually by removing the connection to the main enclosure panel.