Snapmaker/Luban #2364: Bug: xxxx in 3d printing

Bsherba, 2023-09-22:
# 🐞 bug report

### Affected Version(s)

 The issue is caused by version 4.9.0

### Is this a regression? (optional)

 Yes, the previous version in which this bug was not present was: 4.8.2

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error
 The terminal console over wifi for luban 4.9.0 on windows 11 no longer gives feedback from the machine (snapmaker 2 a350).

It doesn't even say "OK" like it did before wifi console was supported.4.8.2 was working properly and I updated yesterday and can no longer receive feedback from the console over wifi. 

The machine does honor commands, but just doesn't respond to them.

Exception or Error (optional)
<pre><code>

</code></pre>

Expected behavior

The machine should be giving me responses, so I can troubleshoot. Trying to work on some linear module issues and the agent wanted me to give them status of the limit switches

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: Windows 11
- Printer: Snapmaker 2 A350, single extruder, new linear modules, no enclosure at this time.

</code></pre>

parachvte, 2023-09-27:
The response via network (machine via Wi-FI) is not a typical Marlin reply (as it does via serial port). It's another protocol used between the communication of machine and Luban. So only "ok" means that the command it received by the machine.

We will soon fix it by providing correct "ok" replies in a minor version (4.9.X).

closed 2023-10-07 by tensile23

Bsherba, 2023-10-07:
Just to ensure you understand the submission entirely,

On 4.9.0 firmware, issuing a command such as M1005 (firmware info macro) has no feedback or response. Not even the "OK" that was previously provided earlier in the snapmaker 2 lifecycle.

The previous firmware, even over wifi, M1005 will provide you the information. My understanding is that originally, this was not possible for the reason you laid out, but one way or another, M1005 does provide the firmware info over wifi connection and has for some time now. 

I am sure you know that, but the comment issued appears to be implying that this isnt the case. It is most likely a misunderstanding on my part. I am not really a programming guy so that's the best i can offer in terms of description of the issue at this time. Kind of a minor miracle i managed to figure out how to submit on github at all 😂 

Thank you for addressing it, as losing the functionality is inconvenient at best.