Snapmaker/Luban #2365: Feature Request: Mark unstable versions of luban as unstable, so that I don't accidentally have my materials wiped and system unusable?

ltmetalman, 2023-09-24:
Would it be possible to just mark lubans that are unstable as unstable? This isn't even the first time this year that all of my snapmaker settings have been completely wiped and that the whole entire software has become unusable. Giving any kind of warning that updating could mean losing everything including functionality, would really be appreciated.

Laser cutter 3d printing time!
[image]

Who needs my material settings when I can get glow in the dark settings for filament I don't have.
[image]

ltmetalman, 2023-09-24:
Its nice at least that downgrading works at getting my settings back... But many of your users likely didn't realize they could do that, so you could also add a "revert to previous version" option somewhere instead.

tensile23, 2023-09-26:
Can you package your config file for us to confirm?
The configuration file folder can be opened with **menu > setting > open config folder** .