Snapmaker/Luban #2366: Bug: Multi-toolpath Rotary Laser doesn't behave as expected.

maihde, 2023-09-25:
# 🐞 bug report

### Affected Version(s)
Only tested in 4.9.0

### To Reproduce
Steps to reproduce the behavior:
1. Start a rotary laser project
2. Add text (I used a single letter M)
3. Create Fill Toolpath on text with movement mode line vertical
4. Create Fill Toolpath on _same_ text with movement mode line horizontal
5. Create Fill Toolpath on _same_ text with movement mode line diagonal

This will render as expected in preview mode, but when actually executed after the first pass the laser no longer turns off correctly as seen below.

[image]

Expected behavior

This should behave identical to making three passes over the same object.

## 🌍  Your Environment

**Platform:**
- Operating System: Windows 11
- Printer: SnapMaker A350

tensile23, 2023-09-26:
A350 been upgraded to the latest firmware version? In older firmware we expected incorrect behavior with gcode generated with 4.9.

maihde, 2023-09-27:
@tensile23 thanks, it was reporting the latest firmware, but I went ahead and clicked the update button and it seemed to update the module and now things appear to be working.  I apologize if there was something in the change log that indicated this incompatibility.  While I'm happy, I'm wondering if there would be a way to warn people of this incompatibility when they connect Luban to the Snapmaker. I appreciate the assistance and you can close this issue.

tensile23, 2023-10-07:
After some new features were added to the firmware(Processing-related), the compatibility issue became serious. We will add firmware version detection on Luban.

closed 2023-10-07 by tensile23