Snapmaker/Luban #2380: Bug: 20/40 Watt Laser and "red laser cross"

Obicom, 2023-10-14:
🐞 bug report

### Affected Version(s)
Luban 4.9.1
20/40 Watt laser
latest firmware

### To Reproduce
Problem 1: In USB mode (serial connection) it is not possible to send a gcode file from Luban, from the workspace, to the laser. Neither as a file nor to start the job directly from there. This only works in WiFi mode. But even there it is only possible to send the file to the controller and not to start the job directly from the workspace.

2nd problem: The red laser cross can only be switched on or is switched on when the gcode file is sent to the controller and the Work Origin is started there via the controller box menu..

Expected behavior
In my opinion, the red laser cross should be turned on, for the 20/40 watt laser, when you manually turn on the laser via Luban or the controller. Is there a gcode command for this that could be saved as a macro?

## 🌍  Your Environment
Windows 11
Luban 4.9.140Watt laser module

Wynand-S, 2024-03-11:
I agree that the 'red laser cross' should be implemented with much better integration.
1. allow the cross to me activated/deactivated from the jogging menu.
2. If I laser a file on the snapmaker without using Luban, the workflow asks for the material height and does not allow me to jog the Z height to see more clearly where the Red cross actually is.  I'm forced to try and align by peering under the laser head which with my low table makes it difficult, and likely also results in a parallax error. 

Further more my A250 always defaults to 130mm/130mm homing position. Even if I correctly jog and set my work coordinates with LUBAN, when I transfer the file to my A250 the coordinates I left my Snapmaker is briefly displayed and then overridden with a default 130x130mm home position.
I prefer to start with my previously defined coordinates and frankly hate that I have to keep jogging back to my preferred coordinates. It's especially frustrating when I want to simply duplicate the previous project.

Is it just me?

theevl, 2024-03-13:
It's not just you.

Implementing the cross hairs from the controller and being able to choose
origin would be very helpful.