Snapmaker/Luban #2415: Bug: Dual extruder auto bed levelling does not know about quick swap kit

fhorlaville, 2023-11-10:
# 🐞 bug report

### Affected Version(s)
firmware v.1.18.0_20231003

### To Reproduce
Steps to reproduce the behavior:
1. Install quick swap kit
2. Install dual extruder 3DP module
3. Calibrate
4. During the auto-bed-levelling the first points are taken outside of the printing table

Expected behavior

Either the quick swap kit bed should be advanced forward by the width of the module quick swap adapter (my preferred option, but that would need a v2 of the kit and also different mounting brackets for the enclosure)

or

The bed-levelling code should be modified to take into account the position of the head with the quick-swap kit installed 

## 🌍  Your Environment

**Platform:**
<pre><code>
- Printer: Snapmaker A250T + quick swap kit + dual extruder

</code></pre>

fhorlaville, 2023-11-11:
Apologies, this should have gone to Snapmaker2-Controller Project - could an admin move it there? Thanks

MIIB1, 2024-02-12:
I think same problem with A350 the auto leveling not working well.

Thanks