Snapmaker/Luban #2435: Bug: Snapmaker Original no longer connects via Serial

wattwood, 2023-11-23:
# 🐞 bug report
Luban fails serial connection to Snapmaker Original running both firmware 2.9 and 2.11.  
Items tested: 
- Swapped PC's
- Swapped USB cable
- Swapped "Brain" (have 2)
- Run as standard user or as Administrator
- Confirmed USB-SERIAL CH340 driver visible in Device Manager Ports

### Affected Version(s)
Fails to connect in versions: 4.10.1, 4.9.1, 4.9.0
Succeeds to connect in version: 4.0.3, 4.2.2, 4.6.1, 4.8.0, 4.8.2

This is a regression for serial connectivity to the Snapmaker Original 3-in-1

### To Reproduce
Steps to reproduce the behavior:
1. Choose 3d Printing
2. Choose Workspace
3. Select COM port
4. Click Connect
5.  Forever spinning where "Connect" text was on button

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System:Windows 11 Home, version 23H2, build 22631.2715
- Printer: Snapmaker Original

</code></pre>

JXGA, 2023-11-28:
Same issue on MacOS. 
Reverting to 4.8.2 also fixes.

BernhardPollerspoeck, 2023-12-21:
Thanks for reporting the issue. Hope some reaction will come soon.

eld3em, 2024-01-09:
Can also replicate this issue, Luban’s logs report:

`error controller:Marlin Error opening serial port “COM4/text”: [Error: Opening COM4: Access denied]`

I’ve tried in different COM ports too, issue persists. Reverting to 4.8.2 resolves, as per previous comments. I note that on the Luban download page, the original snapmaker’s no longer listed…