Snapmaker/Luban #2437: Bug: edit supports not working in 3d printing 4.10.1

mrjdiez, 2023-11-23:
# 🐞 bug report

### Affected Version(s)
The issue is caused by version 4.10.1

### Is this a regression? (optional)

Yes, the previous version in which this bug was not present was: 4.9.1, I updated from this version to the latest

### To Reproduce
Steps to reproduce the behavior:
1. Go to 3d Printing
2. Click on Edit Supports
3. Click on Edit all or Edit with an object selected
4. Paint supports for the model epected
5. Once done click done
6. The GUI shows a progress bar
7. There are no supports in the model. 

Expected behavior
If this was working the model would show supports in the parts painted and not on nothing else

## 🌍  Your Environment

**Platform:** 
<pre><code>
- Operating System: [Windows 10 x64]
- Printer: [Snapmaker Original, Snapmaker A250]

</code></pre>

StinoF, 2023-12-22:
I have the same issue
It either no support or all the supports.

Before the update everything was ok. 

- Operating System: [Windows 10 x64] And [Windows 11 x64]
- Printer: [Snapmaker Original]

markweselmann, 2024-01-10:
Same here, support deletion not working either. 6 weeks since OP, no acknowledgement? Post isn't categorised under 'Type: Bug/Bug Fix' label...

le-avion, 2024-02-16:
I hate to say ditto but since there's no response from developers I will add my own comment. Deleting support doesn't work. The support keeps coming back when generating the G-Code.