Snapmaker/Luban #2460: Bug: arm64 build requires rosetta to be installed in 3d printing

oizone, 2024-02-04:
# 🐞 bug report

### Affected Version(s)
 The issue is caused by version 4.10.2

### Is this a regression?

Not working with older versions either

### To Reproduce
Steps to reproduce the behavior:
1. Using macos arm64 without rosetta installed
2. Go to '3d printing'
3. Click on 'open stl'
4. UI hangs at 75%

server.log error
<pre><code>
2024-02-04T10:52:41.588Z - error uncaughtException: spawn Unknown system error -86
Error: spawn Unknown system error -86
    at ChildProcess.spawn (node:internal/child_process:415:11)
    at Object.spawn (node:child_process:707:9)
    at Slicer.slice (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1014398)
    at Slicer.end (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1015201)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:852808
    at handleCheckModel (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:853114)
    at t.value [as channelMiddleware] (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194915)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:196160
    at Socket.<anonymous> (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194347)
    at Socket.emit (node:events:394:28)
    at Socket.emitUntyped (/Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/typed-events.js:69:22)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/socket.js:614:39
    at processTicksAndRejections (node:internal/process/task_queues:78:11)
2024-02-04T10:52:41.588Z - error app uncaught exception Error: spawn Unknown system error -86
    at ChildProcess.spawn (node:internal/child_process:415:11)
    at Object.spawn (node:child_process:707:9)
    at Slicer.slice (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1014398)
    at Slicer.end (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:1015201)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:852808
    at handleCheckModel (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:853114)
    at t.value [as channelMiddleware] (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194915)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:196160
    at Socket.<anonymous> (/Applications/Snapmaker Luban.app/Contents/Resources/app/src/server/index.js:2:194347)
    at Socket.emit (node:events:394:28)
    at Socket.emitUntyped (/Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/typed-events.js:69:22)
    at /Applications/Snapmaker Luban.app/Contents/Resources/app/node_modules/socket.io/dist/socket.js:614:39
    at processTicksAndRejections (node:internal/process/task_queues:78:11) {
  errno: -86,
  code: 'Unknown system error -86',
  syscall: 'spawn'
}
</code></pre>

Expected behavior
Was expecting the stl to be opened in the preview.

The doesn't happen if rosetta is installed, but the build should be arm64 right?

## 🌍  Your Environment

**Platform:**
<pre><code>

- Operating System: macOS Sonoma 14.2.1
- Printer: Snapmaker A350

</code></pre>

neerdoc, 2024-04-09:
I have the exact same issue. I have an M2 mac and just tried building the binary locally, and the same error occurs, so I don't think the problem is that the build is not ARM64 as I do not have Rosetta installed.

There seems to be problems with Ubuntu ARM64 builds as well: #2101 

So I'm wondering if the issue is ARM64 support across the board?

I will try to dig a bit in this issue.

neerdoc, 2024-04-11:
@oizone Maybe you should change the title? The arm64 image is in fact not arm64 as it contains x86_64 binaries that are required for the software to function. "ARM64 build is broken" would perhaps be a more correct title?

neerdoc, 2024-04-11:
@parachvte @zjiefee 
I would love to help out with compiling this for arm64 on MacOS. 

The actual binary that throws the error first is the `cmd: ../../node_modules/@snapmaker/snapmaker-lunar/engine/MacOS/LunarMP modelrepair -v -p -j ` since it has the wrong CPU architecture.

I can only assume that all three files needs to be compiled for arm64 as they are currently x86_64:
- LunarMP
- LunarSlicer
- LunarTPP

oizone, 2024-04-11:
> @oizone Maybe you should change the title? The arm64 image is in fact not arm64 as it contains x86_64 binaries that are required for the software to function. "ARM64 build is broken" would perhaps be a more correct title?

Well, yeah, it's not a arm64 since it has x86_64 code in it, no. But I think the topic already says this, even though a bit indirect. But anyway, I don't think changing the title will fix this :)

neerdoc, 2024-04-12:
Is the npm package from here?
[Snapmaker/LunarSlicer](https://github.com/Snapmaker/LunarSlicer/tree/main
)? I don't find it very clear if this is the case.

neerdoc, 2024-06-30:
This is still an issue in 4.13.0

I also tried reaching out to Snapmaker support to no avail. They simply answered: "Regarding this issue, there are no plans yet as this would require the R&D department to conduct further evaluation." and closed the ticket.

I would love to help out recompiling the missing parts.

trevseb, 2025-01-19:
Still having this issue too. I will try to reach out to Snapmaker too. It's frustrating that my expensive Artisan printer is hobbled by their software.

ITmaze, 2025-01-19:
This issue continues with 4.15.0 - I've been dealing with Snapmaker support since October 2024. They keep telling me that they cannot reproduce this. Yesterday I discovered what you already have, that there are x64 executables inside the application.

It appears that the source code for that is here: https://github.com/Snapmaker/LunarSlicer

I searched through that source tree for 'arm' and 'arm64' and it appears that this code has never been built on arm64 - unless the compiler doesn't need specific instructions to compile under arm64.

I strongly suspect that the actual root cause however is that the electron builder being used to package Luban hasn't been configured properly to output arm code, unless they're just dropping the binaries from LunaSlicer straight into the application.

ITmaze, 2025-04-13:
It appears that there are two recent updates that add/fix arm64:

- https://github.com/Snapmaker/LunarSlicer/commit/5c40f1bab8a36768b836035547879bd4036576a1
- https://github.com/Snapmaker/LunarMP/commit/97025938b61112357fbf1e8af155c831854de255

ITmaze, 2025-04-15:
From Snapmaker support, I have just received updated `Mach-O 64-bit executable arm64` binaries for the following four files which appears to resolve the issue and I can finally use Luban without Rosetta.

- LubanEngine
- LunarSlicer
- LunarTPP
- LunarMP