Snapmaker/Luban #2470: 3D printing is broken in latest release

blair0011, 2024-03-09:
I have 2 gcode files to print the same 3D object one made by version 4.10.2 which works and the other by 4.11.0 which does not extrude filament. The head moves but the code seems to back out the filament to a point where it doesn't touch the heated surface. I tested the nonworking file 3x and got the same failure. Went back to the old file and it is currently printing well. I did a comparison of the 2 files but the number of differences is too much for me to trace the issue too, also I am have not had enough time to explore Gcode programming so I am not sure what code change could have caused this problem.

coen1111, 2024-07-22:
I had the same issue, for me it seems related to having an enclosure connected. Disconnect the cable of the enclosure seems to solve it. Disabling the door detection of the enclosure does not resolve it.

blair0011, 2024-08-05:
I don't have an enclosure, so that is not my issue.

xiwai2, 2024-09-27:
Hello, I hope you can provide these two Gcodes for us to verify and answer your questions, or you can click this link to contact our after-sales service to solve this problem for you.https://snapmaker.formcrafts.com/support-ticket

blair0011, 2024-09-27:
Unfortunately, I have stopped using Luban for the latest version of Cura. This has fixed the issues stated above.